wpilibsuite/allwpilib #7881: Pose estimator doesn't update odometry pose with vision measurements

calcmogul, 2025-03-25:
Step 9 of `PoseEstimator.addVisionMeasurement()` sets the PoseEstimator pose estimate to a vision-compensated version of the latest odometry object pose, but the odometry object's pose is only updated by twists passed to `PoseEstimator.updateWithTime()`. This means the odometry pose is still allowed to drift over time instead of being gradually fixed by vision measurements. In other words, the odometry is only ever getting fixed by one vision measurement instead of a backlog of all of them.

The old implementation applied vision updates to old pose estimates, then replayed odometry updates up to the current time. That was too expensive at the time, so idk if we can fix this properly.

KangarooKoala, 2025-03-25:
`PoseEstimator` doesn't directly expose `m_odometry` without applying the vision correction, so for standard use, are there any issues aside from `PoseEstimator.resetRotation()` and `PoseEstimator.resetTranslation()` not accounting for the vision corrections?

Also, as a point of clarification, when is m_odometry fixed by a vision measurement? I thought m_odometry is never corrected by vision measurements.

calcmogul, 2025-03-25:
The odometry object pose is never corrected by vision measurements. Effectively, the odometry pose is left to drift while a pose delta is applied on top of it based on m_visionUpdates. This may still be mathematically correct, but it's not numerically stable in general because the odometry estimate error is unbounded. I also don't like that we keep a pose around that doesn't actually get used, because it makes the implementation more confusing.

rzblue, 2025-03-25:
> Effectively, the odometry pose is left to drift while a pose delta is applied on top of it based on m_visionUpdates.

Small clarification: odometry poses are only used to calculate deltas since the most recent vision measurement, which are then applied to the fused pose estimate at the most recent vision measurement

calcmogul, 2025-03-25:
The way the code has been abstracted makes that very unclear.