wpilibsuite/allwpilib #7893: [cscore] Video mode FPS / Frame Time are insufficiently precise on macos

jlmcmchl, 2025-04-08:
**Describe the bug**
When CSCore measures Video Modes on cameras, the measured FPS is truncated, leading to a loss of precision.  Some platforms including macosx are incompatible with this loss of precision, leading to a crash when applying video modes to cameras.

In the documented exception, AVCaptureDevice_Tundra expects a precise frame time of `1000000 / 60000240`, but CSCore provides a frame time of `1 / 60`, causing the app to terminate.

**To Reproduce**
Steps to reproduce the behavior:
1. Run Photonvision
2. Attach a Thrifty Cam
3. Attempt to activate the Thrifty Cam

 - Link to code:
 - Video Mode measurement in USBCameraImplObjc: https://github.com/wpilibsuite/allwpilib/blob/main/cscore/src/main/native/objcpp/UsbCameraImplObjc.mm#L314

**Expected behavior**
Photonvision should crash with the following exception:

```
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureDevice_Tundra setActiveVideoMinFrameDuration:] Not supported - Supported ranges: (
    "<AVFrameRateRange: 0x6000017be830 60.00 - 60.00 (1000000 / 60000240 - 1000000 / 60000240)>"
), tried to set maxFrameRate to 60.000000 (1 / 60)'
*** First throw call stack:
(
        0   CoreFoundation                      0x0000000190f69df0 __exceptionPreprocess + 176
        1   libobjc.A.dylib                     0x0000000190a2eb60 objc_exception_throw + 88
        2   AVFCapture                          0x00000001b11b7bcc -[AVCaptureDALDevice _setActiveVideoMinFrameDuration:] + 0
        3   libcscore.dylib                     0x00000001369aa8c4 _ZN2cs17UsbCameraListener5StartEv + 10356
        4   libcscore.dylib                     0x00000001369ab8a8 _ZN2cs17UsbCameraListener5StartEv + 14424
        5   libdispatch.dylib                   0x0000000190c55b2c _dispatch_call_block_and_release + 32
        6   libdispatch.dylib                   0x0000000190c6f85c _dispatch_client_callout + 16
        7   libdispatch.dylib                   0x0000000190c5e350 _dispatch_lane_serial_drain + 740
        8   libdispatch.dylib                   0x0000000190c5ee2c _dispatch_lane_invoke + 388
        9   libdispatch.dylib                   0x0000000190c69264 _dispatch_root_queue_drain_deferred_wlh + 292
        10  libdispatch.dylib                   0x0000000190c68ae8 _dispatch_workloop_worker_thread + 540
        11  libsystem_pthread.dylib             0x0000000190e09e64 _pthread_wqthread + 292
        12  libsystem_pthread.dylib             0x0000000190e08b74 start_wqthread + 8
)
libc++abi: terminating due to uncaught exception of type NSException
```

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: macos Sequoia 15.4
 - Project Information: `v2025.3.2`

mcm001, 2025-04-08:
Skull 💀

spacey-sooty, 2025-04-08:
Skull 💀

ThadHouse, 2025-04-08:
macOS. The gift that keeps on giving. PR's accepted, we'll see if anyone ever has the time to get to this.

PR #7926 (2025-04-30): [cscore] Add UVC Protocol Support for USB Camera Controls on macOS
## Overview
This MR enhances the cscore project on macOS by implementing UVC (USB Video Class) protocol communication for USB cameras, while also fixing a frame rate precision issue in the AVCaptureDevice configuration.

## Key Changes

### 1. UVC Protocol Implementation
- Added UVC protocol support for USB cameras on macOS
- Implemented common camera property controls through UVC protocol:
  - Exposure
  - Brightness
  - White Balance
  - And other standard UVC properties
- The original implementation was insipred from [openpn [... cut]
closed 2025-04-30 by PeterJohnson