wpilibsuite/allwpilib #7896: Command that can interrupt itself

DJ-Laser, 2025-04-10:
**Is your feature request related to a problem? Please describe.**
This season my team and I decided to implement complex sequences, such as aligning and scoring and bind them to button on the driver's controller. There work great until a part of the robot gets unintentionally stuck on the reef. The driver suggested that pressing the button again would rerun the sequence, but since it never finished due to waiting for the robot to reach a position, pressing the button again simply does nothing as the command is already running. 

**Describe the solution you'd like**
A new enum option in InterruptionBehavior or separate enum that can be set to make the command schedule restart the command if it is scheduled while already active.

**Describe alternatives you've considered**
I've taken to using this ugly pattern to schedule a new instance of the same command logic, but it seems like code smell to me, and is not the best in terms of gc.
```java
public Command scoreCancelSelf() {
    return Commands.defer(
        () -> new ScheduleCommand(score(),
        Set.of());
  }

**Additional context**
Another fun snippet:
```java
mainController
        .povLeft()
        .onTrue(Commands.defer(() -> new ScheduleCommand(s.intake.toggleDeploy()), Set.of()));
```
This one allows the driver to interrupt the command that moves and homes the intake

WispySparks, 2025-04-10:
My team also ran into the same issue, implementing something similar to choosing which command to do on the same button press depending on current state and then manually scheduling it

HarryXChen3, 2025-04-11:
Also hit this one today, after thinking about it some more and with some input from @KangarooKoala (thanks!) I settled with this:

```java
final Command command = doSomething();
trigger
        .onTrue(Commands.runOnce(() -> {
            command.cancel();
            command.schedule();
        }));
```

It avoids recreation of the command and will always re-schedule the command, so I think its a decent workaround for now