wpilibsuite/allwpilib #7906: Replace std::system_clock with std::steady_clock

calcmogul, 2025-04-15:
There's several places that should use std::steady_clock for monotonicity. I'm not sure what we should do for thirdparty deps.
```cpp
[tav@myriad allwpilib]$ rg system_clock
thirdparty/googletest/src/googletest/src/gtest.cpp
1161:  // Fall back to the system_clock when building with newlib on a system
1164:  using clock = std::chrono::system_clock;
1176:             std::chrono::system_clock::now() -
1177:             std::chrono::system_clock::from_time_t(0))

wpiutil/src/test/native/cpp/priority_mutex_test.cpp
17:using std::chrono::system_clock;
105:    system_clock::time_point start_time = system_clock::now();
110:      if (system_clock::now() - start_time > std::chrono::milliseconds(50)) {

wpilibc/src/main/native/cpp/Timer.cpp
22:  using std::chrono::system_clock;
25:      duration_cast<duration<double>>(system_clock::now().time_since_epoch())

wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h
31:/// on a clock other than the system_clock, use std::chrono directly.
33:using TimePoint = std::chrono::time_point<std::chrono::system_clock, D>;
37:class UtcClock : public std::chrono::system_clock {};
51:  return system_clock::to_time_t(
52:      time_point_cast<system_clock::time_point::duration>(TP));
58:  return system_clock::to_time_t(time_point<system_clock, seconds>(
66:  return time_point_cast<seconds>(system_clock::from_time_t(T));
73:  return time_point_cast<nanoseconds>(system_clock::from_time_t(T))

upstream_utils/llvm_patches/0007-Remove-format_provider.patch
180:-      system_clock::from_time_t(mktime(&TM)) + nanoseconds(123456789);
182:-      system_clock::from_time_t(mktime(&TM)) + nanoseconds(23456789);

wpiutil/src/main/native/thirdparty/fmtlib/include/fmt/chrono.h
29:      -> std::chrono::time_point<std::chrono::system_clock, Duration> {
285:using sys_time = std::chrono::time_point<std::chrono::system_clock, Duration>;
514:  // Cannot use std::chrono::system_clock::to_time_t since this would first
515:  // require a cast to std::chrono::system_clock::time_point, which could

wpiutil/src/main/native/cpp/timestamp.cpp
44:             std::chrono::system_clock::now().time_since_epoch())

wpimath/src/main/native/thirdparty/sleipnir/include/sleipnir/autodiff/Profiler.hpp
20:  void StartSetup() { m_setupStartTime = std::chrono::system_clock::now(); }
26:    m_setupDuration = std::chrono::system_clock::now() - m_setupStartTime;
32:  void StartSolve() { m_solveStartTime = std::chrono::system_clock::now(); }
39:    auto now = std::chrono::system_clock::now();
71:  std::chrono::system_clock::time_point m_setupStartTime;
76:  std::chrono::system_clock::time_point m_solveStartTime;

wpimath/src/main/native/thirdparty/sleipnir/src/optimization/solver/SQP.cpp
37:  const auto solveStartTime = std::chrono::system_clock::now();
113:  std::chrono::system_clock::time_point iterationsStartTime;
122:      auto solveEndTime = std::chrono::system_clock::now();
172:    iterationsStartTime = std::chrono::system_clock::now();
177:    std::chrono::system_clock::time_point innerIterStartTime;
179:      innerIterStartTime = std::chrono::system_clock::now();
521:    const auto innerIterEndTime = std::chrono::system_clock::now();

wpimath/src/main/native/thirdparty/sleipnir/src/optimization/solver/InteriorPoint.cpp
44:  const auto solveStartTime = std::chrono::system_clock::now();
140:  std::chrono::system_clock::time_point iterationsStartTime;
149:      auto solveEndTime = std::chrono::system_clock::now();
244:    iterationsStartTime = std::chrono::system_clock::now();
249:    std::chrono::system_clock::time_point innerIterStartTime;
251:      innerIterStartTime = std::chrono::system_clock::now();
787:    const auto innerIterEndTime = std::chrono::system_clock::now();
```

KangarooKoala, 2025-04-18:
Just as a note, the [std::steady_clock](https://en.cppreference.com/w/cpp/chrono/steady_clock) docs note that it's most suited for measuring durations. I'm pretty sure that's what most (if not all) of the occurrences of std::system_clock are used for, but it's still worth noting if that's not the case for a few.

KangarooKoala, 2025-04-18:
Any particular reason to target 2027? (There isn't a particular reason to target main for 2026 either, I'm just curious since this shouldn't be an API-breaking change)

calcmogul, 2025-04-18:
It's not an API-breaking change, but it's a behavior-breaking change.

crueter, 2025-06-23:
The number of uses of system_clock that are in WPILib's direct control is relatively low.

<details><summary>patch</summary>
<p>

```patch
diff --git a/wpilibc/src/main/native/cpp/Timer.cpp b/wpilibc/src/main/native/cpp/Timer.cpp
index 912db0a60..c6fd18b2b 100644
--- a/wpilibc/src/main/native/cpp/Timer.cpp
+++ b/wpilibc/src/main/native/cpp/Timer.cpp
@@ -19,10 +19,10 @@ void Wait(units::second_t seconds) {
 units::second_t GetTime() {
   using std::chrono::duration;
   using std::chrono::duration_cast;
-  using std::chrono::system_clock;
+  using std::chrono::steady_clock;
 
   return units::second_t{
-      duration_cast<duration<double>>(system_clock::now().time_since_epoch())
+      duration_cast<duration<double>>(steady_clock::now().time_since_epoch())
           .count()};
 }
 
diff --git a/wpiutil/src/main/native/cpp/timestamp.cpp b/wpiutil/src/main/native/cpp/timestamp.cpp
index 60efd83ef..faed2aa74 100644
--- a/wpiutil/src/main/native/cpp/timestamp.cpp
+++ b/wpiutil/src/main/native/cpp/timestamp.cpp
@@ -41,7 +41,7 @@ static uint64_t time_since_epoch() noexcept {
 #else
   // 1-us intervals
   return std::chrono::duration_cast<std::chrono::microseconds>(
-             std::chrono::system_clock::now().time_since_epoch())
+             std::chrono::steady_clock::now().time_since_epoch())
       .count();
 #endif
 }
diff --git a/wpiutil/src/test/native/cpp/priority_mutex_test.cpp b/wpiutil/src/test/native/cpp/priority_mutex_test.cpp
index 8e2ec3eaf..22e930997 100644
--- a/wpiutil/src/test/native/cpp/priority_mutex_test.cpp
+++ b/wpiutil/src/test/native/cpp/priority_mutex_test.cpp
@@ -14,7 +14,7 @@ namespace wpi {
 
 #ifdef WPI_HAVE_PRIORITY_MUTEX
 
-using std::chrono::system_clock;
+using std::chrono::steady_clock;
 
 // Threading primitive used to notify many threads that a condition is now true.
 // The condition can not be cleared.
@@ -102,12 +102,12 @@ class BusyWaitingThread {
   void operator()() {
     SetProcessorAffinity(0);
     SetThreadRealtimePriorityOrDie(21);
-    system_clock::time_point start_time = system_clock::now();
+    steady_clock::time_point start_time = steady_clock::now();
     m_started.Notify();
     while (m_run.test_and_set()) {
       // Have the busy waiting thread time out after a while.  If it times out,
       // the test failed.
-      if (system_clock::now() - start_time > std::chrono::milliseconds(50)) {
+      if (steady_clock::now() - start_time > std::chrono::milliseconds(50)) {
         return;
       }
     }
```
</p>
</details> 

Outside of this, all the remaining usages are indeed hidden in LLVM, libfmt and gtest. A (not so-)quick grep gives this output:

<details><summary>Grep</summary>
<p>

```cpp
wpiutil/src/main/native/thirdparty/fmtlib/include/fmt/chrono.h:29:      -> std::chrono::time_point<std::chrono::system_clock, Duration> {
wpiutil/src/main/native/thirdparty/fmtlib/include/fmt/chrono.h:285:using sys_time = std::chrono::time_point<std::chrono::system_clock, Duration>;
wpiutil/src/main/native/thirdparty/fmtlib/include/fmt/chrono.h:514:  // Cannot use std::chrono::system_clock::to_time_t since this would first
wpiutil/src/main/native/thirdparty/fmtlib/include/fmt/chrono.h:515:  // require a cast to std::chrono::system_clock::time_point, which could
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:31:/// on a clock other than the system_clock, use std::chrono directly.
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:33:using TimePoint = std::chrono::time_point<std::chrono::system_clock, D>;
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:37:class UtcClock : public std::chrono::system_clock {};
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:51:  return system_clock::to_time_t(
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:52:      time_point_cast<system_clock::time_point::duration>(TP));
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:58:  return system_clock::to_time_t(time_point<system_clock, seconds>(
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:66:  return time_point_cast<seconds>(system_clock::from_time_t(T));
wpiutil/src/main/native/thirdparty/llvm/include/wpi/Chrono.h:73:  return time_point_cast<nanoseconds>(system_clock::from_time_t(T))
thirdparty/googletest/src/googletest/src/gtest.cpp:1161:  // Fall back to the system_clock when building with newlib on a system
thirdparty/googletest/src/googletest/src/gtest.cpp:1164:  using clock = std::chrono::system_clock;
thirdparty/googletest/src/googletest/src/gtest.cpp:1176:             std::chrono::system_clock::now() -
thirdparty/googletest/src/googletest/src/gtest.cpp:1177:             std::chrono::system_clock::from_time_t(0))
upstream_utils/llvm_patches/0007-Remove-format_provider.patch:180:-      system_clock::from_time_t(mktime(&TM)) + nanoseconds(123456789);
upstream_utils/llvm_patches/0007-Remove-format_provider.patch:182:-      system_clock::from_time_t(mktime(&TM)) + nanoseconds(23456789);```
</p>
</details>

sciencewhiz, 2025-07-04:
Closed by #8053

closed 2025-07-04 by sciencewhiz

rzblue, 2025-07-04:
I don't think this was closed by that PR- that PR only changed the behavior of `wpi::now()` on systemcore to not offset the monotonic clock to match the wall clock.

rzblue, 2025-07-04:
Now that I'm looking at the usages listed above, only the usages in `priority_mutex_test` and sleipnir are usages that could use `steady_clock`

- the usage in Google test is for platforms without a monotonic clock
- `frc::GetTime` (in Timer.cpp) is supposed to return the system clock 
- the usage in llvm intentionally uses the system clock
- the fmtlib usage is for formatting `system_clock` types 
- the usage in `timestamp.cpp` is for an initial offset for monotonic timestamps