wpilibsuite/allwpilib #7912: Start log files connected to FMS with event_match

ArchdukeTim, 2025-04-19:
Current logs that are part of a match are named FRC_date_time_eventcode_matchnumber.wpilog

It'd be nice if they were labeled eventcode_matchnumber_frc_date_time.wpilog, so that you could more easily sort them when the drive has log files from non-official matches

PeterJohnson, 2025-04-22:
I think the current system has some advantages, such as making it easy to find "nearby" logs (e.g. what did the test run just prior to the match show?) and is consistent with the way DS logs are sorted.  The other thing this helps with is if for whatever reason the match info didn't make it to the robot code, the log is still fairly easy to find based on date/time.

This might be a better issue for the tooling side; e.g. AdvantageScope or DataLogTool could parse the filenames and give sortable columns to make it easier to find a particular match?

ArchdukeTim, 2025-04-22:
Yeah the timestamps do have that advantage. Perhaps I should ask CTRE to change to match WPILib's format, since they do MATCH_device_date.hoot