wpilibsuite/allwpilib #7917: Rewrite TrapezoidProfile internals to use phase space formulation

calcmogul, 2025-04-20:
This should fix several corner cases and hard-to-reproduce bugs in the current implementation, and make the internals easier to understand and maintain.

calcmogul, 2025-04-25:
Relevant implementation by FRC team 100 based on https://arxiv.org/pdf/2210.01744:

https://github.com/Team100/all25/blob/1d97a30da424a297200504498ff8eb23ce8bb248/lib/src/main/java/org/team100/lib/profile/incremental/TrapezoidProfile100.java

calcmogul, 2025-09-18:
Contents of duplicate issue #8234:

Here's an overview of the trapezoid profile phase space math:
https://underactuated.mit.edu/dp.html#minimum_time_double_integrator

Here's an implementation of the general control policy, though it should calculate the intercept time like ExponentialProfile instead.

[double_integrator_time_optimal_controller.py](https://github.com/user-attachments/files/22414834/double_integrator_time_optimal_controller.py)

Since an exponential profile has the velocity model dx/dt = −kᵥ/kₐx + 1/kₐu and a trapezoid profile has the velocity model dx/dt = 1/kₐu, one could try starting from https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/algorithms.md#closed-form-solution-for-an-exponential-motion-profile and https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/algorithms/ExponentialProfileModel.py and simplifying.

cb144p, 2025-09-18:
One of these bugs can happen when the difference in the goal and current position are small, but differences in velocity are much larger. The Robot.cpp of a Timed Robot Skeleton that demonstrates this is provided below:
```
// Copyright (c) FIRST and other WPILib contributors.
// Open Source Software; you can modify and/or share it under the terms of
// the WPILib BSD license file in the root directory of this project.

#include "Robot.h"
#include <frc/trajectory/TrapezoidProfile.h>
#include <units/length.h>
#include <units/velocity.h>
#include <units/acceleration.h>
#include <frc/smartdashboard/SmartDashboard.h>

frc::TrapezoidProfile<units::meters> profile{frc::TrapezoidProfile<units::meters>::Constraints{4_mps, 5_mps_sq}};
bool done = false;
frc::TrapezoidProfile<units::meters>::State currState{0_m, 3_mps};
frc::TrapezoidProfile<units::meters>::State goalState{.01_m, 0_mps};
units::second_t dt = 0.02_s;

Robot::Robot() {}
void Robot::RobotPeriodic() {}

void Robot::AutonomousInit() {}
void Robot::AutonomousPeriodic() {}

void Robot::TeleopInit() {}
void Robot::TeleopPeriodic() {}

void Robot::DisabledInit() {}
void Robot::DisabledPeriodic() {}

void Robot::TestInit() {
  frc::SmartDashboard::PutNumberArray("State", std::array<const double, 2>{currState.position.value(), currState.velocity.value()});
}

void Robot::TestPeriodic() {
  if (!done) {
    frc::TrapezoidProfile<units::meters>::State next = profile.Calculate(dt, currState, goalState);
    frc::SmartDashboard::PutNumberArray("State", std::array<const double, 2>{currState.position.value(), currState.velocity.value()});
    if (goalState == currState) {
      done = true;
    } else {
      currState = next;
    }
  }
}

void Robot::SimulationInit() {}
void Robot::SimulationPeriodic() {}

#ifndef RUNNING_FRC_TESTS
int main() {
  return frc::StartRobot<Robot>();
}
#endif
```
This code produces the following values as shown in AdvantageScope:
[image]
The profile should look approximately like:
[image]