wpilibsuite/allwpilib #7929: Use `dwz` to reduce the size of debug symbols

qwertychouskie, 2025-04-26:
For example, `libwpimathd.so.debug` shipped in `linuxathena` is 332.8MB before running `dwz` and 316.2MB afterwards.  It's not a ton, but it all adds up with every library/platform combo.

In a perfect world, the debug info also wouldn't be shipped by default, but would be hosted and the libraries set up to use debuginfod so they can be downloaded on-demand by the debugger if/when they are actually needed, but that's more advanced.

calcmogul, 2025-04-26:
The Linux Athena symbols are already split so they don't get deployed to the target platform.

Normally I'd agree with this, but the offline installation case is the reason we include the debug symbols by default.

calcmogul, 2025-04-26:
What would be nice is if we could strip just the Eigen symbols, since that's where all the bloat comes from.

closed 2026-08-24 by Gold856