wpilibsuite/allwpilib #7945: Units: Measures are a bit weird to create

qwertychouskie, 2025-04-29:
The current way to create a distance in inches:
```java
Distance wheelDiameter = Inches.of(6);
```
"Inches of 6" doesn't read naturally, making this code a bit awkward.  Changing `of` to `from` was suggested, but although this is slightly better, it's still a bit weird.

Something like one of these would feel more natural:
```java
Distance wheelDiameter = Unit.Inches(6);     // Unit would contain all valid types of units
Distance wheelDiameter = Distance.Inches(6); // Distance would contain valid types of distance units
```
This way, you can also use `var`, since the created type will be pretty clear, and you won't have to remember which exact class contains the unit you want:
```java
var wheelDiameter = Unit.Inches(6);        // wheelDiameter will be of the type Distance
var kMaxSpeed = Unit.MetersPerSecond(2.5); // wheelDiameter will be of the type LinearVelocity
```

PeterJohnson, 2025-04-29:
The proposed approach isn’t naturally extensible to user-created units though, as users can’t inject new unit static functions into the Units class.

qwertychouskie, 2025-04-29:
> The proposed approach isn’t naturally extensible to user-created units though, as users can’t inject new unit static functions into the Units class.

You could create a `MyUnits` class that extends `Unit` and implements your custom units.  Alternatively, you could just use the old syntax anywhere you create custom units, there's no reason that the approaches can't co-exist.  That said, in what circumstances would custom units be needed in the first place?  There are only so many units that exist in the world, unless I'm missing something.

SamCarlberg, 2025-08-01:
`of` is a fairly standard factory function name in the Java ecosystem (see `List.of`, `Set.of`, `Map.of`, ...). The ideal would be UDLs (like C++) or extension methods (like Kotlin) to make measure creation read more like English (`Inches.of(1)` -> `1_inch` or `1.inch`), but we have to use the tools the language gives us.

If we define unit-specific factory functions, they should be defined in the relevant unit class. Instead of `Unit.Inches(x)`, use `Distance.inches(x)`. Keep in mind that these kinds of functions can only be available for predefined unit types; custom compound units like volts per second (used for slew rate controls and SysID) and inches per degree (encoder conversion factors) couldn't have predefined factories; otherwise, we'd have to maintain a combinatorial explosion of factories for all the compound units.

Mixing these factory patterns harms readability and code comprehension, since _some_ blessed units would have dedicated factories and everything else would have to use the unit-provided `of` factories.

```java
// using unit-specific factories
var target = Distance.inches(42);

// custom units can't have factories
var other = FTCTileLength.of(1);

// nor can compound units
var kP = Volts.per(Second).of(2.2);
```