wpilibsuite/allwpilib #7974: wpiutil fails to link

OhmV-IR, 2025-05-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout v2025.3.2
2. cmake --preset default -DBUILD_SHARED_LIBS=OFF -DWITH_SIMULATION_MODULES=OFF -DWITH_TESTS=OFF
3. cd build-cmake
4. cmake --build . --parallel 4
5. cd lib
6. cp *.a /2702-Vision/vcpkg_installed/x64-linux/lib
7. make

Makefile I am using:
```
MAKEFLAGS := --jobs=$(shell nproc)
#MAKEFLAGS +=--output-sync=target

OBJS=object.o
CXXFLAGS+=--std=c++20
CXXFLAGS+=-I../../lib 
CXXFLAGS+=-DDISPLAY=0
CXXFLAGS+=-DCAMSERVER=0
CXXFLAGS+=-g
CXXFLAGS+=-Ofast
CXXFLAGS+=$(shell pkg-config --cflags ../../vcpkg_installed/x64-linux/lib/pkgconfig/opencv4.pc)
CXXFLAGS+=-I$(shell pwd)/../../vcpkg_installed/x64-linux/include/ntcore
CXXFLAGS+=-I$(shell pwd)/../../vcpkg_installed/x64-linux/include/wpiutil
CXXFLAGS+=-I$(shell pwd)/../../vcpkg_installed/x64-linux/include/cameraserver
CXXFLAGS+=-I$(shell pwd)/../../vcpkg_installed/x64-linux/include/cscore
CXXFLAGS+=-I$(shell pwd)/../../vcpkg_installed/x64-linux/include/wpimath
CXXFLAGS+=$(shell pkg-config --cflags ../../vcpkg_installed/x64-linux/lib/pkgconfig/libcamera.pc)
CXXFLAGS+=$(shell pkg-config --cflags ../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc)
CXXFLAGS+=$(shell pkg-config --cflags ../../vcpkg_installed/x64-linux/lib/pkgconfig/gtk+-x11-3.0.pc)
CXXFLAGS+=$(shell pkg-config --cflags ../../vcpkg_installed/x64-linux/lib/pkgconfig/libxml-2.0.pc)
CXXFLAGS+=$(shell pkg-config --cflags ../../vcpkg_installed/x64-linux/lib/pkgconfig/bzip2.pc)
LDFLAGS+=-L../../lib
LDFLAGS+=-L$(shell pwd)/../../vcpkg_installed/x64-linux/lib
LIBS+=-lntcore -lcscore -lcameraserver -lwpiutil -lwpimath -lfmt -lwpinet -lwpilibNewCommands -lwpiHal -lwpilibc -luv -lxrpVendordep -lromiVendordep -lapriltag -lglass -lglassnt -lwpigui -lglfw3 -limgui
LIBS+=-lprotobuf
LIBS+=$(shell pkg-config -libs ../../vcpkg_installed/x64-linux/lib/pkgconfig/libcamera.pc)
LIBS+=$(shell pkg-config -libs ../../vcpkg_installed/x64-linux/lib/pkgconfig/opencv4.pc)
LIBS+=$(shell pkg-config -libs ../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc)
LIBS+=$(shell pkg-config -libs ../../vcpkg_installed/x64-linux/lib/pkgconfig/gtk+-x11-3.0.pc)
LIBS+=-lxml2 -lexpat
LIBS+=$(shell pkg-config -libs ../../vcpkg_installed/x64-linux/lib/pkgconfig/bzip2.pc)
all: object replay nettableserver testnetworktablesclient udppacketstest calibrate testudpserver logtostdvideo

object: $(OBJS) ../../lib/lib2702.a
	$(CXX) $(CXXFLAGS) $(LDFLAGS) -o object $(OBJS) -l2702 $(LIBS)

replay: replay.o ../../lib/lib2702.a
	$(CXX) $(CXXFLAGS) $(LDFLAGS) -o replay replay.o -l2702 $(LIBS)
```
Console output:
```
make replay
Package ../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc was not found in the pkg-config search path.
Perhaps you should add the directory containing `../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc.pc'
to the PKG_CONFIG_PATH environment variable
Package '../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc', required by 'virtual:world', not found
Package ../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc was not found in the pkg-config search path.
Perhaps you should add the directory containing `../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc.pc'
to the PKG_CONFIG_PATH environment variable
Package '../../vcpkg_installed/x64-linux/lib/pkgconfig/OpenEXR.pc', required by 'virtual:world', not found
g++ --std=c++20 -I../../lib  -DDISPLAY=0 -DCAMSERVER=0 -g -Ofast -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/opencv4  -I/2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/include/ntcore -I/2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/include/wpiutil -I/2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/include/cameraserver -I/2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/include/cscore -I/2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/include/wpimath -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/libcamera   -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/gtk-3.0 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/pango-1.0 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/glib-2.0 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../lib/glib-2.0/include -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/fribidi -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libpng16 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/cairo -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/pixman-1 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/gdk-pixbuf-2.0 -I/usr/include/x86_64-linux-gnu -I/usr/include/webp -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/gio-unix-2.0 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/atk-1.0 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/at-spi2-atk/2.0 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/dbus-1.0 -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../lib/dbus-1.0/include -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/at-spi-2.0 -pthread  -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include/libxml2 -DLIBXML_STATIC  -I../../vcpkg_installed/x64-linux/lib/pkgconfig/../../include  -L../../lib -L/2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib -o replay replay.o -l2702 -lntcore -lcscore -lcameraserver -lwpiutil -lwpimath -lfmt -lwpinet -lwpilibNewCommands -lwpiHal -lwpilibc -luv -lxrpVendordep -lromiVendordep -lapriltag -lglass -lglassnt -lwpigui -lglfw3 -limgui -lprotobuf -L../../vcpkg_installed/x64-linux/lib/pkgconfig/../../lib -lcamera -lcamera-base  -L../../vcpkg_installed/x64-linux/lib/pkgconfig/../../lib -lopencv_gapi4 -lopencv_stitching4 -lopencv_alphamat4 -lopencv_aruco4 -lopencv_bgsegm4 -lopencv_bioinspired4 -lopencv_ccalib4 -lopencv_dnn_objdetect4 -lopencv_dnn_superres4 -lopencv_dpm4 -lopencv_face4 -lopencv_freetype4 -lopencv_fuzzy4 -lopencv_hdf4 -lopencv_hfs4 -lopencv_img_hash4 -lopencv_intensity_transform4 -lopencv_line_descriptor4 -lopencv_mcc4 -lopencv_rapid4 -lopencv_reg4 -lopencv_saliency4 -lopencv_signal4 -lopencv_stereo4 -lopencv_structured_light4 -lopencv_phase_unwrapping4 -lopencv_superres4 -lopencv_optflow4 -lopencv_surface_matching4 -lopencv_tracking4 -lopencv_highgui4 -lopencv_datasets4 -lopencv_text4 -lopencv_plot4 -lopencv_videostab4 -lopencv_videoio4 -lopencv_wechat_qrcode4 -lopencv_xfeatures2d4 -lopencv_shape4 -lopencv_ml4 -lopencv_ximgproc4 -lopencv_video4 -lopencv_xobjdetect4 -lopencv_objdetect4 -lopencv_calib3d4 -lopencv_imgcodecs4 -lopencv_features2d4 -lopencv_dnn4 -lopencv_flann4 -lopencv_xphoto4 -lopencv_photo4 -lopencv_imgproc4 -lopencv_core4 -L../../vcpkg_installed/x64-linux/lib/pkgconfig/../../lib/manual-link/opencv4_thirdparty -lippiw -lippicv -lprotobuf -lade -lGL -lGLU -lwebpdecoder -lwebp -lwebpdemux -lwebpmux -lsharpyuv -ltiff -llzma -ljpeg -ljxl -ljxl_cms -llcms2 -ljxl_threads -lbrotlienc -lhwy -lopenjp2 -lquirc -L/usr/lib/gcc/x86_64-linux-gnu/13 -lgomp -lpthread -lfreetype -lbz2 -lpng16 -lz -lbrotlidec -lbrotlicommon -lharfbuzz -lhdf5 -ldl -lm -lrt   -L../../vcpkg_installed/x64-linux/lib/pkgconfig/../../lib -lgtk-3 -lgdk-3 -lm -lpangocairo-1.0 -lm -lm -lm -lm -lXrandr -lXfixes -lX11 -lcairo-gobject -lgdk_pixbuf-2.0 -ljpeg -ltiff -latk-bridge-2.0 -latspi -ldbus-1 -lXtst -lXi -latk-1.0 -Wl,--export-dynamic -lgmodule-2.0 -lepoxy -lGL -lEGL -lpangoft2-1.0 -lpango-1.0 -lm -lfribidi -lharfbuzz -lm -lstdc++ -lcairo -ldl -lpng16 -lfontconfig -lfreetype -lXext -lXrender -lX11 -lpixman-1 -lgio-2.0 -lgobject-2.0 -lffi -lgmodule-2.0 -lglib-2.0 -latomic -lm -pthread -lpcre2-8 -lz  -lxml2 -lexpat -L../../vcpkg_installed/x64-linux/lib/pkgconfig/../../lib -lbz2 
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libcscore.a(Frame.cpp.o): in function `cs::Frame::ConvertYUYVToBGR(cs::Image*)':
/home/ohm/Downloads/allwpilib/cscore/src/main/native/cpp/Frame.cpp:436:(.text+0xd3c): undefined reference to `cv::cvtColor(cv::_InputArray const&, cv::_OutputArray const&, int, int)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libcscore.a(Frame.cpp.o): in function `cs::Frame::ConvertYUYVToGray(cs::Image*)':
[... cut ...]
/usr/bin/ld: /home/ohm/Downloads/allwpilib/wpinet/src/main/native/cpp/WebSocket.cpp:151:(.text+0x1b00): undefined reference to `wpi::SHA1::Update(std::basic_string_view<char, std::char_traits<char> >)'
/usr/bin/ld: /home/ohm/Downloads/allwpilib/wpinet/src/main/native/cpp/WebSocket.cpp:153:(.text+0x1b1b): undefined reference to `wpi::SHA1::RawFinal(wpi::SmallVectorImpl<char>&)'
```

**Expected behavior**
Links successfully

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04 in a VM

**Additional context**
Linker links other things fine like ntcore, just wpiutil is causing problems for me.

sciencewhiz, 2025-05-13:
> LIBS+=-lntcore -lcscore -lcameraserver -lwpiutil -lwpimath -lfmt -lwpinet -lwpilibNewCommands -lwpiHal -lwpilibc -luv -lxrpVendordep -lromiVendordep -lapriltag -lglass -lglassnt -lwpigui -lglfw3 -limgui

That doesn't look like the right order, compared to https://github.com/wpilibsuite/allwpilib/blob/main/MavenArtifacts.md#provided-artifacts

OhmV-IR, 2025-05-13:
Have updated to this and stopped building gui since I don't need it: LIBS+=-lwpiutil -lwpimath -lwpinet -lntcore -lcscore -lcameraserver -lwpilibc -lwpilibNewCommands -lapriltag which follows that order, but still getting these errors:
```
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ntcore_cpp_types.cpp.o): in function `nt::GetString(unsigned int, wpi::SmallVectorImpl<char>&, std::basic_string_view<char, std::char_traits<char> >)':
ntcore_cpp_types.cpp:(.text+0x1139): undefined reference to `wpi::SmallVectorBase::grow_pod(void*, unsigned long, unsigned long)'
/usr/bin/ld: ntcore_cpp_types.cpp:(.text+0x11c9): undefined reference to `wpi::SmallVectorBase::grow_pod(void*, unsigned long, unsigned long)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ntcore_cpp_types.cpp.o): in function `nt::GetAtomicString(unsigned int, wpi::SmallVectorImpl<char>&, std::basic_string_view<char, std::char_traits<char> >)':
ntcore_cpp_types.cpp:(.text+0x12f0): undefined reference to `wpi::SmallVectorBase::grow_pod(void*, unsigned long, unsigned long)'
/usr/bin/ld: ntcore_cpp_types.cpp:(.text+0x1388): undefined reference to `wpi::SmallVectorBase::grow_pod(void*, unsigned long, unsigned long)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ntcore_cpp_types.cpp.o): in function `nt::GetRaw(unsigned int, wpi::SmallVectorImpl<unsigned char>&, std::span<unsigned char const, 18446744073709551615ul>)':
ntcore_cpp_types.cpp:(.text+0x17c8): undefined reference to `wpi::SmallVectorBase::grow_pod(void*, unsigned long, unsigned long)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ntcore_cpp_types.cpp.o):ntcore_cpp_types.cpp:(.text+0x1850): more undefined references to `wpi::SmallVectorBase::grow_pod(void*, unsigned long, unsigned long)' follow
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(InstanceImpl.cpp.o): in function `nt::ListenerStorage::~ListenerStorage()':
InstanceImpl.cpp:(.text._ZN2nt15ListenerStorageD2Ev[_ZN2nt15ListenerStorageD5Ev]+0x2f): undefined reference to `wpi::detail::SafeThreadOwnerBase::~SafeThreadOwnerBase()'
/usr/bin/ld: InstanceImpl.cpp:(.text._ZN2nt15ListenerStorageD2Ev[_ZN2nt15ListenerStorageD5Ev]+0x391): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: InstanceImpl.cpp:(.text._ZN2nt15ListenerStorageD2Ev[_ZN2nt15ListenerStorageD5Ev]+0x3a1): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(InstanceImpl.cpp.o): in function `nt::local::StorageImpl::~StorageImpl()':
InstanceImpl.cpp:(.text._ZN2nt5local11StorageImplD2Ev[_ZN2nt5local11StorageImplD5Ev]+0xe6): undefined reference to `wpi::deallocate_buffer(void*, unsigned long, unsigned long)'
/usr/bin/ld: InstanceImpl.cpp:(.text._ZN2nt5local11StorageImplD2Ev[_ZN2nt5local11StorageImplD5Ev]+0x336): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: InstanceImpl.cpp:(.text._ZN2nt5local11StorageImplD2Ev[_ZN2nt5local11StorageImplD5Ev]+0x359): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: InstanceImpl.cpp:(.text._ZN2nt5local11StorageImplD2Ev[_ZN2nt5local11StorageImplD5Ev]+0x570): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: InstanceImpl.cpp:(.text._ZN2nt5local11StorageImplD2Ev[_ZN2nt5local11StorageImplD5Ev]+0x5b1): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: InstanceImpl.cpp:(.text._ZN2nt5local11StorageImplD2Ev[_ZN2nt5local11StorageImplD5Ev]+0x7cf): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ListenerStorage.cpp.o): in function `nt::ListenerStorage::Activate(unsigned int, unsigned int, std::function<bool (unsigned int, nt::Event*)>)':
ListenerStorage.cpp:(.text+0x24d): undefined reference to `wpi::SmallVectorBase::mallocForGrow(void*, unsigned long, unsigned long, unsigned long&)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ListenerStorage.cpp.o): in function `nt::ListenerStorage::DoAddListener(unsigned int)':
ListenerStorage.cpp:(.text+0xbfa): undefined reference to `wpi::CreateSignalObject(unsigned int, bool, bool)'
/usr/bin/ld: ListenerStorage.cpp:(.text+0xdc9): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: ListenerStorage.cpp:(.text+0xdd9): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ListenerStorage.cpp.o): in function `nt::ListenerStorage::Reset()':
ListenerStorage.cpp:(.text+0x11f4): undefined reference to `wpi::detail::SafeThreadOwnerBase::operator bool() const'
/usr/bin/ld: ListenerStorage.cpp:(.text+0x1200): undefined reference to `wpi::detail::SafeThreadOwnerBase::Stop()'
/usr/bin/ld: ListenerStorage.cpp:(.text+0x12e1): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: ListenerStorage.cpp:(.text+0x12f1): undefined reference to `wpi::DestroySignalObject(unsigned int)'
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ListenerStorage.cpp.o): in function `nt::ListenerStorage::NotifyTimeSync(std::span<unsigned int const, 18446744073709551615ul>, unsigned int, long, long, bool)':
[... cut ...]
/usr/bin/ld: /2702-Vision/2025/Object/../../vcpkg_installed/x64-linux/lib/libntcore.a(ListenerStorage.cpp.o): in function `nt::ListenerStorage::Notify(unsigned int, unsigned int, std::basic_string_view<char, std::char_traits<char> >, unsigned int, std::basic_string_view<char, std::char_traits<char> >)':
ListenerStorage.cpp:(.text+0xa870): undefined reference to `wpi::SetSignalObject(unsigned int)'
/usr/bin/ld: ListenerStorage.cpp:(.text+0xa87e): undefined reference to `wpi::SetSignalObject(unsigned int)'
```

ThadHouse, 2025-05-13:
That order is backwards. With static libraries, you need to link from the highest level library to the lowest level library.

closed 2025-05-13 by OhmV-IR