wpilibsuite/allwpilib #8041: Timestamps seed to system clock

ThadHouse, 2025-06-27:
Currently the timestamps on systemcore seed to system clock. We probably don't want this, and should instead just use monotonic clock directly without any seeding.

AustinSchuh, 2025-06-27:
Consider switching to `clock_gettime(CLOCK_MONOTONIC, ...)` too.  That'll provide nanosecond accurate time.

FYI, a lot of the kernel syscalls really want to use `CLOCK_MONOTONIC` not `CLOCK_MONOTONIC_RAW`.  I've gone down that path in the past, ended up backing out and just going with `CLOCK_MONOTONIC`.

(Feel free to steal anything you like from https://github.com/RealtimeRoboticsGroup/aos/blob/main/aos/time/time.cc )

ThadHouse, 2025-06-27:
Is C++ steady_clock generally driven by CLOCK_MONOTONIC? That way teams could use either? And you'd just get NS precision if you use the C function?

AustinSchuh, 2025-06-27:
https://github.com/llvm-mirror/libcxx/blob/master/src/chrono.cpp#L217

https://github.com/gcc-mirror/gcc/blob/master/libstdc++-v3/src/c++11/chrono.cc#L87

Looks like it.

ThadHouse, 2025-06-27:
> How much overlap is there between this and [#7906](https://github.com/wpilibsuite/allwpilib/issues/7906)?

None. Different parts of the code

commit 0304f501 (2025-07-04): [wpiutil] Use steady_clock directly on SystemCore (#8053)

Don’t offset it from program start time.

This will make it much easier to coordinate timestamps across the system.

Closes #8041
closed 2025-07-08 by rzblue