wpilibsuite/allwpilib #8054: avoid field-relative ChassisSpeeds

truher, 2025-07-02:
I'm poking at 2027 and noticed the reworking of the way ChassisSpeeds handles the "field relative" notion.

Could I suggest an alternative?  I think it would be better for the data type to indicate the thing being measured -- use ChassisSpeeds for robot-relative, and some new type for field-relative.  Meaningless like this should be impossible:

```
new ChassisSpeeds(0,0,0).toRobotRelative(0).toRobotRelative(0);
```

eh?

calcmogul, 2025-07-02:
We'd potentially need to double the function overloads in wpimath to handle both.

truher, 2025-07-02:
Would you though?  Are you thinking of functions like toSwerveModuleStates()?  I think it would be ok for those functions to only take robot-relative input (which is what they implicitly do now anyway), no?

truher, 2025-07-03:
Oh, so the normal pattern is to pass around a boolean?  Maybe just stick that boolean inside this class then?