wpilibsuite/allwpilib #8215: [java] SIGSEGV from race conditions between Preferences.setNetworkTableInstance() and NetworkTableInstance.close()

kcooney, 2025-09-06:
**Describe the bug**
We have tests for code that uses Preferences where we create a new NetworkTableInstance per test. The tests use a JUnit Rule ([source code](https://github.com/Prospect-Robotics/lib2813/blob/9de1625265a268975de847c2964df326d72dcb65/lib/src/test/java/com/team2813/lib2813/preferences/IsolatedPreferences.java)) that creates a unique NetworkTableInstance instance before each test is run.

As we have added more of these tests, we were increasingly getting test failures in GitHub workflows due to the JVM dying with a SIGSEGV.

The tests consistently pass for me when running locally via Gradle.

The relevant lines of the JVM error file appear to be:

```
Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
J 1843  edu.wpi.first.networktables.NetworkTablesJNI.setTopicPersistent(IZ)V (0 bytes) @ 0x00007ff978d647ec [0x00007ff978d647a0+0x000000000000004c]
j  edu.wpi.first.networktables.Topic.setPersistent(Z)V+5
j  edu.wpi.first.wpilibj.Preferences.lambda$setNetworkTableInstance$0(Ledu/wpi/first/networktables/NetworkTableEvent;)V+38
j  edu.wpi.first.wpilibj.Preferences$$Lambda$491+0x00007ff9141a5208.accept(Ljava/lang/Object;)V+4
J 1835 c1 edu.wpi.first.networktables.NetworkTableInstance$ListenerStorage.lambda$startThread$0()V (327 bytes) @ 0x00007ff971574684 [0x00007ff971573a80+0x0000000000000c04]
j  edu.wpi.first.networktables.NetworkTableInstance$ListenerStorage$$Lambda$492+0x00007ff9141a5660.run()V+4
j  java.lang.Thread.run()V+11 java.base@17.0.16
v  ~StubRoutines::call_stub
```

I believe this is due to the `NetworkTableListener` callback installed by `Preferences.setNetworkTableInstance()` being called after the `NetworkTableInstance` was closed.

We "solved" this by adding a small delay between calling `Preferences.setNetworkTableInstance(NetworkTableInstance.getDefault())` and closing the previously-set `NetworkTableInstance` ([code](https://github.com/Prospect-Robotics/lib2813/blob/9998ee2ef1036b2d4ccd76f2e10c47d6b83b9eb2/lib/src/test/java/com/team2813/lib2813/preferences/IsolatedPreferences.java#L34-L35)).

Links
- [GitHub action logs](https://github.com/Prospect-Robotics/lib2813/actions/runs/17458522539/job/49732832933) (open the "Archive test artifacts" step to download the JVM logs)
- [commit with our "fix"](https://github.com/Prospect-Robotics/lib2813/pull/51/commits/6269fac67d4df0de67403aa5e4a14a7491dfab63)

**Desktop (please complete the following information):**
 - OS: MacOS 15.5 (note: issue does not reproduce locally)
 - Project Information: 

```

[hs_err_pid2310.log](https://github.com/user-attachments/files/22183848/hs_err_pid2310.log)

WPILib Information:
Project Version: 2025.3.1
VS Code Version: 1.96.2
WPILib Extension Version: 2025.2.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.43.1
Java Debug Extension Version: 0.58.2
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: /Users/kevincooney/wpilib/2025/jdk
Vendor Libraries:
    PathplannerLib (2025.2.1)
    CTRE-Phoenix (v5) (5.35.1)
    CTRE-Phoenix (v6) (25.2.1)
    REVLib (2025.0.1)
    WPILib-New-Commands (1.0.0)
    photonlib (v2025.3.2)
```

PeterJohnson, 2025-09-06:
Understood.  I’m just explaining the underlying issue to guide the approach to fixing it.

kcooney, 2025-09-06:
For those following along, a less janky work-around is to call `NetworkTableInstance.waitForListenerQueue()` before `NetworkTableInstance.close()` if the network potentially has/had listeners (as would be the case if it was passed to `Preferences.setNetworkTableInstance()`)

kcooney, 2025-09-14:
@PeterJohnson I have a possible fix: #8226

closed 2026-10-04 by PeterJohnson