wpilibsuite/allwpilib #8233: Detailed Trigger

JosephTLockwood, 2025-09-18:
**Is your feature request related to a problem? Please describe.**
One issue our team faced last year was the lack of clear visibility into when and what parts of the trigger were (or weren't) being triggered, without having to delve into multiple subsystems and graph them in AScope.

**Describe the solution you'd like**
I would like to have an option to create a more detailed trigger that logs everything together and provides clear visibility, all in one section, on what conditions have or haven't been met for the trigger. 

**Describe alternatives you've considered**
Logging values in RobotContainer with paths set for each Trigger but this really bloated up the code.

**Additional context**
Here is a rough pass on something that does have minimal functionality and does work. However, this still needs a significant update, and I'm unsure about the next steps. I initially tried making a Sendable; this worked with `SmartDashboard.putData()`, however, it didn't work at all when testing with Epilogue. I ended up scrapping Sendable and decided to get this to work with Epilogue with a `CustomLoggerFor`, which is very cumbersome.
[DetailedTrigger.zip](https://github.com/user-attachments/files/22411411/DetailedTrigger.zip)

Example code and how it looks in code and on AdvantageScope
```
    exampleTrigger =
        new DetailedTrigger(
            () -> arm.getPosition().isNear(arm.getTargetPosition(), arm.getTolerance()),
            new String[] {"position", "target", "tolerance"},
            () -> arm.getPosition().in(Degrees),
            () -> arm.getTargetPosition().in(Degrees),
            () -> arm.getTolerance().in(Degrees));
```

[image]

GGreenix, 2025-09-20:
That could be nice if implemented, I think in general, in commands as well, see logs of when commands are executed in conditions and in sequence, Sometimes you think you set up the conditions right, but requirements or conditions aren't right, seeing it visually could help.

SamCarlberg, 2025-09-22:
> @SamCarlberg, do you have any ideas on how this implementation could work better with Epilogue?

The data that the trigger depends on should already be logged; it seems wasteful and redundant to include it again under the trigger. Maybe new annotations could be introduced to mark triggers (or any logged item, really) as dependent on other items to include metadata for data analysis tools like AdvantageScope to use

For example, a class like this:
```java
@Logged
class Arm {
  @DependsOn("getPosition()")
  @DependsOn("getTargetPosition()")
  @DependsOn("getTolerance()")
  public final Trigger exampleTrigger =
    new Trigger(() -> getPosition().isNear(getTargetPosition(), getTolerance()));

  public Angle getPosition() { ... }
  public Angle getTargetPosition() { ... }
  public Angle getTolerance() { ... }
}
```

could generate logging calls like this:

```java
public class ArmLogger extends ClassSpecificLogger<Arm> {
  @Override
  public void update(EpilogueBackend backend, Arm object) {
    if (Epilogue.shouldLog(Logged.Importance.DEBUG)) {
      backend.log("Example Trigger", object.exampleTrigger.getAsBoolean(), "Position", "Target Position", "Tolerance");
      backend.log("Position", object.getPosition());
      backend.log("Target Position", object.getTargetPosition());
      backend.log("Tolerance", object.getTolerance());
    }
  }
}
```